elephantrobotics/myCobot #9: Issue with compatibility of Arduino software

cpilon, 2021-01-24:
I have tried running the latest Arduino examples (f187422451a775a8d4eb4ef234972f653b22aafc) with my cobot and they do not seem to work. The code compiles and transfers to the M5Stack Basic properly by the robot does not move. I had to roll back to the Arduino examples from 07/12/2020 (**8f155a7cf3e0e6bbc610be2a5340391da960b0b7**) to be able to run the examples. 

Also, I noticed the myCobotV1.3 does not work with my cobot. Only myCobotV1.2 works for the Calibration and MainControl (like the other Arduino examples, everything compiles and transfers for the robot doesn't move).

Can you please advise? Should I be able to use the latest code on any version of myCobot?

wxy6655, 2021-01-27:
Sorry, you can download the latest firmware burner now，His name is MyStudio，The download link is：‘https://github.com/elephantrobotics/myStudio/releases/tag/v0.0.4’

cpilon, 2021-03-31:
Hi, I downloaded the latest version of MyStudio (V1.2) and I still have the issue.
I can only make the calibration and "MainControl" work with the old myCobotV1.2 application.
I also tried the Transponder with the python API examples and they also do not work for me. 

Is there someone with older versions of myCobot that are not compatible with your new software developments?

wxy6655, 2021-04-01:
Hi, I'm sorry that we have stopped updating the version of MycobotV1.3. If you need to run the latest history and need to update the firmware to Atom, you can link to MyStudio to update the firmware to Atom. The latest firmware is compatible with all the history.

cpilon, 2021-04-01:
Hi, thanks for the reply but I think you are missing my point.
I want to use MyStudio to update the firmware of the M5Stack Basic, but when I do, none of the "Tools" work after I flash them.

Is MyStudio V1.2 supposed to work with ALL myCobots?

Thanks,
Christian

wxy6655, 2021-04-01:
You need to do a firmware update to the atom，Basic is used to burn the sample programs in the 'Tools' project. The Atom on the end is the control center. You must use MyStudio to update the Atom on the end to the latest version of the firmware.

cpilon, 2021-04-01:
Ok, right. You are correct it seems the Atom version was causing the issue. Thanks!

wxy6655, 2021-04-01:
You're welcome. Enjoy your use

closed 2021-04-01 by wxy6655