elephantrobotics/myCobot #17: AtomMain 2.8.1 has broken Encoder Values

peteh, 2021-05-29:
In 2.8.1 Atom Main, the encoder values are not updating or just wrong. 

The following code will not show correct encoder values: 
for (int i = 0; i < 6; i++)
```
{
    angles.joint_angle[i] = m_mycobotBasic.getEncoder(i + 1);
}
```

It can also be reproduced by flashing the latest MainControl app into Basic and trying to record and replay a motion. The replaying will not work and move to random poses instead of the expected track. 

The bug seems to have been introduced with 2.8.1. Version 2.8 is working properly.

zlj-zz, 2021-06-01:
Please try Atom2.8

peteh, 2021-06-03:
That is more reliable but also sometimes gives me -1 for encoder values. 

I currently workaround it by trying to read the value up to 3 times which is more robust. However, this interferes with timing.

zlj-zz, 2021-06-03:
The reason for `-1` is that the serial port does not read the data, which may be read too fast. So that the serial port can't react.
If you are reading data continuously, you can skip them when you get `[]` or `-1`.

zlj-zz, 2021-06-04:
I am sorry， maybe I didn't find out. where you get  the `-1`?

```
m_mycobotBasic.getEncoder(i + 1);
```
here?

peteh, 2021-06-04:
Yes, this function seems to return -1.

zlj-zz, 2021-06-04:
OK. The reason is that valid data cannot be read. I think you just need to skip `-1`.

API will only throw out the possible existence of wrong data, and the user will deal with this situation.😊

eacousineau, 2021-06-25:
Possibly relates https://github.com/elephantrobotics/pymycobot/issues/17