elephantrobotics/myCobot #18: How to read button on Atom

peteh, 2021-05-30:
Hi, 

I could not find any documentation how to read the button on the Atom Lite. 

It's supposed to be connected to G39 on the Atom, however the following code does not work when flashed to the basic: 
`int value = m_mycobotBasic.getDigitalInput(39);`

it always returns -1. 

How can I get button Events?

wxy6655, 2021-06-30:
Hello, the function of this case is not open yet, only IO pins are open at present

peteh, 2021-07-02:
Is there a plan when it will be available? Currently the button is pretty useless.

wxy6655, 2021-07-05:
Hello, I have confirmed that the button function is available, provided that you set it to the input state, I see that you seem to use it directly, maybe you can set it to the input state first, you can use set_pin_mode() to set it , And then use the read function.

zhouliqingyannuo, 2021-07-05:
unsubscribe

 

 
&mdash;

peteh, 2021-07-05:
> 
> 
> Hello, I have confirmed that the button function is available, provided that you set it to the input state, I see that you seem to use it directly, maybe you can set it to the input state first, you can use set_pin_mode() to set it , And then use the read function.

I tried to do that, however without success (tested with Atom 2.8). 

Do you have some sample code that I could use? Do I have to constantly poll the state of the input or is it possible to react on events?

wxy6655, 2021-07-06:
```C
#include <MycobotBasic.h>
#include <ParameterList.h>

MycobotBasic myCobot;

void setup(){
  myCobot.setup();                                      
  myCobot.powerOn();                                
  pinMode(21,OUTPUT);
  pinMode(22,OUTPUT);
  myCobot.setPinMode(39,INPUT);
}

void loop(){
  Serial.println(myCobot.getDigitalInput(39));
  delay(200);
}
```