elephantrobotics/myCobot #25: mycobot_24px.h EncodeRange error: Can't build examples Calibration, miniRoboFlow, myPalletizerFlow, MainControl

thandal, 2021-09-17:
For those four examples, I get the following error:

In file included from C:\Users\XXX\Documents\Arduino\libraries\MycobotBasic/MycobotLanguage.h:7:0,
                 from C:\Users\XXX\Documents\Arduino\libraries\MyCobotBasic\examples\Calibration\Calibration.ino:3:
C:\Users\XXX\Documents\Arduino\libraries\MycobotBasic/mycobot_24px.h:1467:7: error: 'EncodeRange' does not name a type
 const EncodeRange mycobot_24pxEncodeRange[] PROGMEM = {

Yuzhiqi11, 2021-09-18:
For this issue, maybe need check this file C:\Users\XXX\Documents\Arduino\libraries\M5Stack\src\utility/ln_eSPI_Setup.h

Yuzhiqi11, 2021-09-18:
For this issue, pls check C:\Users\XXX\Documents\Arduino\libraries\M5Stack\src\utility/ln_eSPI_Setup.h to uncomment the last line "//#define USE_M5_FONT_CREATOR"
For more details pls check FAQ in https://github.com/elephantrobotics/myCobot/blob/main/Arduino/README.md, hope it wll be helpful to you~

thandal, 2021-09-18:
Thank you! I had seen that note, but somehow didn't associate it with this problem!

This lets me build Calibration.ino. But I still can't build miniFlow, myPalletizerFlow, or MainControl, which fails with the error:
'class SMSBL' has no member 'Feedback' -- which I think is related to https://github.com/elephantrobotics/myCobot/issues/24 -- I must not have quite the right version for the library :(

Yuzhiqi11, 2021-09-22:
I'm sorry that our Arduino library for SCServo temporarily is not open for public, but for the issue you mentioned, we update our code which may help you. Pls check https://github.com/elephantrobotics/myCobot/tree/main/Arduino/MyCobotBasic/examples/miniRoboFlow and  https://github.com/elephantrobotics/myCobot/tree/main/Arduino/MyCobotBasic/examples/MyPalletizerRoboFlow.