elephantrobotics/myCobot #33: Adaptive gripper not working with Arduino

jdk-21, 2021-11-12:
I'm trying to test the adaptive gripper with the Arduino example, but the gripper won't move.
```CPP
/// 此文件用于mycobot设备控制末端夹爪运动，对应烧录设备为Basic
/// 烧录前请将夹爪张开至最大角度

#include <MyCobotBasic.h>
#include <ParameterList.h>

MyCobotBasic myCobot;   
 
void setup() {
  myCobot.setup();
  delay(10);
  myCobot.powerOn();
  delay(10);
  myCobot.setGripperIni();                          //设定夹爪当前位置电位值为2048
  delay(20);
  Serial.begin(9600);
}

void loop() {
    myCobot.setEncoder(7,2048);                     //设定末端夹爪打开至2048电位值
    delay(2000);
    Serial.println(myCobot.getGripperValue());      //读取末端夹爪电位值
    delay(50);
    myCobot.setEncoder(7,1700);                     //设定末端夹爪打开至1700电位值
    delay(2000);
    Serial.println(myCobot.getGripperValue());      //读取末端夹爪电位值
    delay(50);
}
```
I completley opened the gripper and than burned the image as instructed.
After the transfer has finished nothing happens. The gripper stays open. I can check in the serial monitor that the gripper is completely open (prints 255). If I close the gripper manually the value gets smaller. So the `getGripperValue()` method seems to work. But the gripper doesn't move.

I'm using the latest Atom firmware. I burned it with the myCobot burning tool.
The gripper I'm using is the adaptive gripper.

Eheech, 2021-11-15:
Have you updated the firmwares of Atom and Basic? And which version of mycobot are you using ? Have you changed baud in MyCobotBasic.h ? And please upload screenshot of arduino compile and upload codes , thank you very much .

jdk-21, 2021-11-18:
Thank you for the replay. You were right. I had an old version of AtomMain installed. I updated Atom to V4.0.1 and know the gripper is working.

closed 2021-11-18 by jdk-21