elephantrobotics/myCobot #39: array library not found

skywo1f, 2022-10-31:
When attempting to compile any examples for the arduino, I get:

In file included from C:\Users\iviti\sketchbook\libraries\MyCobotBasic/MyCobotSaver.h:5:0,
                 from C:\Users\iviti\sketchbook\libraries\MyCobotBasic/MyCobotBasic.h:7,
                 from C:\Users\iviti\sketchbook\mySetRGB\mySetRGB.ino:1:
C:\Users\iviti\sketchbook\libraries\MyCobotBasic/ParameterList.h:27:10: fatal error: array: No such file or directory
 #include <array>
          ^~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

ferrolho, 2024-03-25:
Hey! Did you end up solving this issue? I think this is related to the _avr-libstdcpp_ 
 mentioned in [10.3 Use of the Arduino library](https://docs.elephantrobotics.com/docs/gitbook-en/10-ArduinoEnv/10.3-arduinolib_use.html).

ferrolho, 2024-03-26:
Okay, I've managed to fix this by following the instructions in Chinese [here](https://github.com/elephantrobotics/MyCobotBasic?tab=readme-ov-file#%E7%BC%96%E8%AF%91%E5%89%8D%E7%9A%84%E4%BF%AE%E6%94%B9). It requires copying and pasting the `lib\avr-libstdcpp` folder included with the [MyCobotBasic](https://github.com/elephantrobotics/MyCobotBasic) repository into your computer's `Arduino/libraries` folder.

But now I am getting another error:
```
In file included from /Users/henrique/Documents/Arduino/libraries/avr-libstdcpp/new_handler.cc:26:0:
/Users/henrique/Documents/Arduino/libraries/avr-libstdcpp/new:40:10: fatal error: stddef: No such file or directory
 #include <stddef>
          ^~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

didmou, 2024-10-28:
Hello, I had exactly the same problem, to solve it,  have use the IDE Arduino 1.8.19 !!! And NOT IDE Arduino 2.3.3 !!! 
I don't know why exactly there is this difference... but if it's can help ...

etlcook, 2024-11-08:
> Hello, I had exactly the same problem, to solve it, have use the IDE Arduino 1.8.19 !!! And NOT IDE Arduino 2.3.3 !!! I don't know why exactly there is this difference... but if it's can help ...

Thank you for this comment, this solved the problem for me!