elephantrobotics/myCobot #48: myCobot 280 Pi Firmware Issue - Requires Reflashing After Power Cycle

Tareq-Mh, 2024-06-16:
#### Description:
We've purchased a myCobot 280 Pi and updated the Atom board to version 6.4. However, we are experiencing a recurring issue where the robot stops working after each power cycle. This requires us to manually reflash the Atom board firmware from the Raspberry Pi to restore functionality.

#### Problem:
- After each power cycle, the motors do not respond to commands (e.g., `send_angles([0,0,50,0,0,0],100)` has no effect).

#### Temporary Solution:
- Manually reflashing the Atom board firmware using the myStudio app on the Raspberry Pi restores functionality. However, this needs to be repeated every time the power is cut, which is highly inconvenient.

#### Steps to Reproduce:
0. Update the Atom board firmware to 6.4.
1. Power cycle the myCobot 280 Pi.
2. Observe that the motors no longer respond to commands.
3. Manually reflash the Atom board firmware using the myStudio app on the Raspberry Pi.
4. The robot starts working again.

#### Request:
Could you please provide a permanent solution or suggest a method to avoid having to reflash the Atom board after every power cycle?

Thank you for your assistance!

always-lgtm, 2024-09-06:
How did you manage this? When i launch myStudio on the rpi, mine says no board detected and does not even allow me to flash.

sgbaird, 2024-09-25:
> * Manually reflashing the Atom board firmware using the myStudio app on the Raspberry Pi restores functionality. However, this needs to be repeated every time the power is cut, which is highly inconvenient.

A colleague, @gursi26, seems to be experiencing the same issue and can reproduce the behavior.