elephantrobotics/myCobot #51: MyArm M750 Robot device regarding burning and updating firmware.

nanditha-10, 2024-12-18:
I followed the procedure outlined in the official documentation for burning and updating firmware using Mystudio. I downloaded all the required controllers (M5Stack Basic, Atom, and Pico) and performed the flashing process to burn the firmware. The burning process was indicated as successful.

However, during the verification step:

**The Connect Test for Atom displays NO instead of OK.**
[image]

**The 7 Move Servos test also shows NO instead of OK.**
[image]

**For reference, I followed the steps provided in the official documentation here:**
https://docs.elephantrobotics.com/docs/myarm-master_750-en/4-FunctionsAndApplications/5-BasicFunctions/5.2-SoftwareUsageInstructions/5.2.3-flash_firmwares.html

Kindly assist in resolving this issue.

closed 2025-01-13 by nanditha-10

nanditha-10, 2025-01-13:
[image]
[image]

Resloved , Thankyou

phongtran218, 2026-10-03:
Hi @nanditha-10 , I know this issue had been resolved long ago, however, can I ask how did you resolve it ? I am currently facing the same issue